riscv-non-isa/riscv-nexus-trace #5: Missing LICENSE file

jjscheel, 2021-06-04:
While the [README](https://github.com/riscv/tg-nexus-trace/blob/master/README.md) file references the project is licensed under the CC-BY-4.0 license and provides a link to the full license text. the lack of a LICENSE file prevent GitHub from recognizing the license.

I propose adding a plain text [LICENSE](https://github.com/riscv/riscv-isa-manual/blob/master/LICENSE) file like the one used in the ISA manual.

I'm happy to add this into the project directly and will do so in the coming days. If anyone has concerns, please raise them.

Thanks,
-Jeff

RobertChyla, 2021-06-04:
Thanks - I see you do similar things in other TG pages, so this is good to unify it.

jjscheel, 2021-06-07:
Committed directly to master.  Closing.

closed 2021-06-07 by jjscheel