riscv-non-isa/riscv-nexus-trace #39: Feedback on RISC-V Trace Connectors Specification

ved-rivos, 2023-11-18:
1. "MIPI Debug & Trace Connector Recommendations" white paper (dated 7/2/2021) defines two pin options - "Narrow JTAG + Trace" and "JTAG + Trace" - for the 20-pin connector. This pin mapping supports the use of pin 14/16/18/20 as TRC_DATA[0/1/2/3] and pin 12 as TRC_CLK. It is unclear why Chapter 2 indicates that these are new additions to MIPI. 
2. Chapter 2, Table 1 - shows two pins as SerialTrace. What is the function of having two SerialTrace pins and why do they have identical names?
3. Chapter 2, Table 1 - The layout in Table 1 does not match the mappings in Table 4 of the MIPI white paper. Specifically the JTAG mapping shown as mapping B in Table 1 is deprecated by MIPI. The Table 1 should be reconciled with MIPI and specifically identify why chapter 2 should diverge from MIPI recommendation.
4. The option of providing power through pin 11/13 is noted by MIPI whitepaper in Table 4 as supported by ARM but not supported by MIPI. The rationale for RISC-V to support this is not clear.
5. Please provide more rationale for chapter 3 not simply pointing to Table 11 of MIPI white paper i.e. the need for Chapter 3.
6. Please provide the rationale for chapter 4 not being a reference to chapter 7 of the MIPI white paper. 

Please provide further arguements for why this document should not be a white paper (mostly reference ARM and MIPI specifications where appropriate) instead of a normative RISC-V specification.

RobertChyla, 2023-12-08:
@ved-rivos 

Two commits:

https://github.com/riscv-non-isa/tg-nexus-trace/commit/7eea6ee594bef5d8cf3e19cd4fb77e8e67337f9a
https://github.com/riscv-non-isa/tg-nexus-trace/commit/98fb6b2ca6fe3a180e2da32a9f80290ecb12830a

are IMO addressing all issues mentioned above (Google DOC comments provide more explanations).

Corresponding PDF (1.0.0_rc10, 2023-12-08) is here:

https://github.com/riscv-non-isa/tg-nexus-trace/actions/runs/7146467079

I assume after you verify it is OK, you will close this issue.

mipsrobert, 2024-03-09:
I am closing all Connectors PDF related issues with same comments as all issues were handled via comments/discussions in SINGLE Google Docs. Relevant links are as follows:

Notes to Connectors PDF:   https://docs.google.com/document/d/1iNbB7-nTiQQ4vQBzLqxtBDQxT51QeUQMd9MFOe0jsBs

Connectors PDF rc20:       https://github.com/riscv-non-isa/tg-nexus-trace/commit/221f6b1cef94c5f503554e0cb0deb4046dd49686
Connectors for ARC review: https://github.com/riscv-non-isa/tg-nexus-trace/commit/1de77dcff1a75197232f72b8fe6863892da0cc74

closed 2024-03-09 by mipsrobert