riscv-non-isa/riscv-nexus-trace #40: Section 3.4 - PIB Idle cycles

ved-rivos, 2023-11-18:
The section 3.4 indicates that idle sequences may be transmitted between packets and the idle sequences may be a sequence of `n` where `n` may be 0 or any integer. One key aspect that is not specified is that for the probe to be able to synchronize with the bit stream, the encoder must provide a sequence of 8 idle bits to be able to disambiguate a packet boundary. These 8 idle bits may not occur naturally in the trace stream and may need to be explicitly inserted periodically. Guidance is needed on the need for these synchronizing sequences and the periodicity at which they should be generated if they dont occur naturally.

RobertChyla, 2023-11-27:
E-Trace just added 'Encapsulation Protocol'. For N-Trace idle is defined as 0xFF, which CANNOT appear inside of normal N-Trace message (0xFF may appear as last byte of message, but it is OK). I will look at this topic again.

mipsrobert, 2024-03-09:
This was addresses in related chapter in N-Trace PDF (as this is specific to different protocols).

mipsrobert, 2024-03-09:
I am closing all Connectors PDF related issues with same comments as all issues were handled via comments/discussions in SINGLE Google Docs. Relevant links are as follows:

Notes to Connectors PDF:   https://docs.google.com/document/d/1iNbB7-nTiQQ4vQBzLqxtBDQxT51QeUQMd9MFOe0jsBs

Connectors PDF rc20:       https://github.com/riscv-non-isa/tg-nexus-trace/commit/221f6b1cef94c5f503554e0cb0deb4046dd49686
Connectors for ARC review: https://github.com/riscv-non-isa/tg-nexus-trace/commit/1de77dcff1a75197232f72b8fe6863892da0cc74

closed 2024-03-09 by mipsrobert