riscv-non-isa/riscv-nexus-trace #42: Trace example `t1` makes non-standard use of `IndirectBranchHist` Message

albertschulz, 2024-01-16:
The [t1 trace](https://github.com/riscv-non-isa/tg-nexus-trace/blob/master/refcode/c/examples/t1/output/test-NEX.bin) contains multiple messages indicating a indirect branch, but actually being generated for direct, conditional branches.

For example MSG #2:
```
. 0x70 011100_00: TCODE[6]=28 (MSG #2) - IndirectBranchHist
. 0x10 000100_00: BTYPE[2]=0x0
. 0x55 010101_01: ICNT[10]=0x151 (337)
. 0x45 010001_01: UADDR[6]=0x11 (17)
. 0x00 000000_00:
. 0x00 000000_00:
. 0x00 000000_00:
. 0x00 000000_00:
. 0x00 000000_00:
. 0x0B 000010_11: HIST[36]=0x80000000 (-2147483648)
. next_iaddr=0x200102bc, EmitICNT(n=337,hist=0x80000000)
```

Before this message, the trace is decoded until PC `0x200102b8`.

The [disassembly](https://github.com/riscv-non-isa/tg-nexus-trace/blob/master/refcode/c/examples/t1/test-OBJD.txt#L227) for this address looks as follows:
```
  200102b2:	fc442703          	lw	a4,-60(s0)
  200102b6:	87c2                	mv	a5,a6
> 200102b8:	00f71363          	bne	a4,a5,200102be <xrle_compress+0x136>
```

Problem: `bne` is not an indirect jump.

For me, that seems like a kind of sync message, or somehow replacing the nexus resource-full message type, since always the HIST field is completely used.

Maybe I'm missing something here or have the wrong understanding. Does somebody could provide some clarification, would be awesome.

Best,
Albert

mipsrobert, 2024-01-19:
During ARC review issue to better clarify Synchronization messages came up. We will review all parts related to SYNC messages.

mipsrobert, 2024-03-07:
Hi @albertschulz - we are finalizing the spec (after ARC review notes) and I am looking at all other 'pending' issues.

This is indeed sync-related. These are 'dropped' when HIST is full. Example encoder is not implementing ResourceFull. I will added it. When HIST overflows, then we cannot issue 'Sequention Instruction Counter' - it means, that Sequential instruction counter MUST be ONLY used in BTM mode. This will be addressed in the spec.

And reference code will be updated as well. It will allow:
1. Work in HTM mode (when 'resource-full' with RCODE=1 MUST be generated).
2. Work in BTM mode (potential 'resource-full' when small I-CNT will be used).
3. Work in BTM mode (potential 'SYNC=Sequential Instruction Counter" on small I-CNT).

I simply 'abandoned' any reference code changes until the spec is ratified (I have better version on my private repo ...).
The ambition is to have it 'fully-compliant' and provide options to use different methods and field-sizes. For example forcing I-CNT to be 3-bit only one may generate trace with a lot of overflows.

mipsrobert, 2024-03-07:
Just more notes ... I will close this issue once PDF is 'fully compliant'. Changes to Reference Code will come later (I will work on it in a background while spec will be in 'public review' phase (hopefully soon!).

mipsrobert, 2024-03-30:
N-Trace PDF 1.0.0_rc30 (after second round of ARC reviews) should have this clarified.
Reference code will be updated later (after all 3 PDFs are Frozen).

closed 2024-03-30 by mipsrobert