riscv-non-isa/riscv-nexus-trace #53: Feedback on RISC-V Trace Control Interface Specification - chapter 6

ved-rivos, 2024-04-13:
1. The number of bits requiring reset should be reduced to the minimal needed to ensure that the trace tool can initialize the trace encoder to desired state. For trace encoder, Reset to 0 should  be needed only for trTeActive and trTeEnable. When trTeEnable is 0 the trace encoder does not emit any trace and all other register state may then be initialized by the trace  tools to the desired state. Read-only fields by definition have a defined read-only value. This feedback applies to all trace control registers.

2. trTeInstTrigEnable - The description incorrectly states that this signal is set or cleared by the Debug Module instead of the Trigger Module specified by the Sdtrig ISA extension. The description can also be reworded for better readability/clarity.

3.  trTeInstTracing - this signal is volatile and can be set or cleared by triggers. Guideline for a safe sequence to write this bit - for instance as part of writing say 1 to trTeInstStallOrOverflow is missing in the specification. Likewise for trTeDataTracing.

4. trTeInstSyncMax - the periodicity allowed by this field  - 16 through 256 - seems quite limited and may lead to high frequency synchronization on certain high performance implementations - for instance a 8 wide implementation operating at 4 GHz configured to count instruction halfwords may need a synchronization every 16 cycles i.e. 4ns. Is such high frequency synchronization intended?

5. trTeInstExtendAddrMSB - the name of the control bit and its description is misleading as the encoder does not do any extension of address bits. Further it is indicated as being limited to Sv39/48/57 - which contradicts N-trace which allows compression for Sv32 and also physical addresses.

6. trTeDataDrop - why is this bit not redundant in relation to trTeDataStallOrOverflow

7. trTeDataStallOrOverflow - why is this bit set when a overflow message is generated and not when overflow is detected? Why is there a subtle difference in behavior compared to trTeInstStallOrOverflow?

8. trTeDataDropEna - how is this control different than trTeDataStallEna?

9. Specify behavior if both trTeDataNoValue and trTeDataNoAddr are both set to 1.

10. trTeDataAddrCompress - For E-trace data trace,  the address is always differential and the data can be compressed using XOR or differential encoding. The 4 encodings specified here do not match up with the 4 `diff` options specified by E-trace data trace protocol.

12. "In a system with Internal Core Timestamp counter" - the earlier description implied that the core level sharing was only the clock and the  timestamp counter is per hart. This text seems to contradict that. Having a shared timestamp counter among multiple harts is not very useful - especially since it may be reset by per-hart controls. This should be changed to a per hart timestamp counter.

13. trTsCount - this field should be WARL as it is legal to hardwire this unit to 0 when an  internal timestamp unit is not implemented.

14. Timestamp unit - two terms are used "Internal timestamp" and "Internal core timestamp". Are these intended  to be different. If not please unify to use a single term.

15. trTsRunInDebug - this should be renamed "trTsRunWhenHalted" to support additional cases where the timestamp counter may halt  - such as low power entry, being held in reset, or due to security reasons as being discussed in context of external debug security.

16. trTsPrescale - is the prescale to scale up or scale down?

17. Timestamp units - bits  23:16 - does "System-dependent" mean they are designated for custom extension?

18. Section 6.2.1 - should be renamed to "Trigger Module"

19. section 6.2.1  - (breakpoint or watchpoint) - is this an exhaustive list? Why is trace-on/trace-off action prohibited for other forms of triggers?

jgamoneda, 2024-04-22:
Planning on updating the spec for the follow issues:
2,4,11,12,16

Response:
1. RC: I really want to have trace functional with minimal settings. Just enable and it reasonably works. Trace may be enabled (for debug ...) from early boot code. Setting many registers would complicate things. Also with WARL semantics it should not be beneficial to have reset value as the only value.

2. Jay: Suggest rewording to: “Allows trTeInstTracing to be set or cleared by Trace-on and Trace-off triggers signals generated by the corresponding Trigger module”. **ACTION UPDATE SPEC.**

3. RC: When trace triggers are used to enable trace you do not set this bit. Otherwise you set it. Setting in the middle while triggers are on/off will be random. The only reasonable use case would be to enable tracing to see where code is if from some reason enable trigger didn't fire

4. Robert: Description was not fixed after we removed from Ntrace. Jay: Agree, we need to align this control bit to the updated description used in the message document. **ACTION UPDATE SPEC**.

5. Jay: This bit is used to notify of dropped data message. This may occur independent of a stall and/or overflow.

6. Jay: To me it makes sense to get set when the message is generated. Makes sure the last stall/overflow is detected when the message is sent. Doing it earlier has no real value.

7. RC: Stall and drop are different things. If someone doesn't want to stall they may still want to drop data trace to prevent loosing of instruction trace.

8. Robert: You may only see that transfer occurred and have timestamp (if configured). With heave filtering it can be seen as generating markers which can be counted. Setting data trace at Ethernet controller register may allow you to see who and how often is sending packets ...

9. RC. It was heavily discussed. E-trace is not perfect ...

10. Rc. Terms hart/core are not always consistently used.

11. Jay :We will update DOC to WARL **ACTION UPDATE SPEC**

12. Jay: We will update DOC to unify to a single term. **ACTION UPDATE SPEC**

13. RC. Seems you are mixing halt of hart with halt of timestamp. This bit is strictly for debug mode

14. RC. It is divider so down ...

15. Rc. No it's just different for different systems.... Jay: Should we rename this to Vendor defined?

16. Agree it should be renamed to Trigger Module **ACTION UPDATE SPEC**

17. ??, Not clear ont he comment, Trace-on/off is only defined for trigger actions 2/3 when supported by the hart

mipsrobert, 2024-05-02:
See comment above - closing it.

closed 2024-05-02 by mipsrobert

ved-rivos, 2024-05-14:
Reopening this issue for a few open points:

1. The reset burden should be removed to the minimal needed for SW
     such as the external trace tools to be able to configure the trace 
     control interface to enable. Besides the fields that would prevent 
     the trace tool from performing such configurations, all other fields
     should have a legal but unspecified value at reset. The documented
     sequence for enabling reset, discovery, and enabling trace. This 
     includes actions such training PIB sink or configuring base addresses
    in system memory for the RAM sinks. 

2. trTeInstTracing - the specific question was with respect to
    trTeInstStallOrOverflow which are RW1C. Consider a condition such as
    follows:
         - trace-on sets trTeInstTracing to 1
         - A stall occurs causing trTeInstStallOrOverflow to be set to 1
         - The tracing tool reads trTeControl - let this read value be A.
            The read value A has trTeInstTracing as 1.
         - A trace-off trigger clears trTeInstTracing to 0 in trTeControl
         - The tracing tool now set trTeInstStallOrOverflow in A and 
            writes this value to trTeControl. This will cause the
            trTeInstTracing  to be set to 1 when it should be 0.
      Since the trTeInstTracing has two writers - a hardware writer
      and a SW writer - a safe sequence to writing trTeControl when
      trace is enabled should be documented. This may be to disable
      trace before making any writes to trTeControl.

3. trTeDataStallOrOverflow - Is there a reason why the behavior
    for asserting this is different than behavior specified for 
    trTeInstStallOrOverflow. Jay indicates that this should be
    done when Overflow message is generated and not when Overflow
    condition is detected then the trTeInstStallOrOverflow behavior 
    should be updated to match trTeDataStallOrOverflow. It may however
    be more useful to indicate on overflow condition being detected than
    when it is removed (When Overflow message is generated) as it allows
    to provide a status indication that trace has stopped due to  an overflow.
    Delaying it to when the overflow message is generated i.e. overflow 
    condition no longer exists may not be useful as the overflow message 
    itself will tell the decoder that such a condition occurred.

4. trTeDataAddrCompress. The E-trace specification provides the following:
            00: Full address and data (sync)
            01: Differential address, XOR-compressed data
            10: Differential address, full data
            11: Differential address, differential data
     The only two address compression options are to send full address or 
     to send differential data. The is an option for XOR compression of data 
     but not the address. 
     The trace control specification indicates:
           0: Only send full (unmodified) addresses
           1: Use XOR compression
           2: Use differential compression
           3: Dynamically select XOR or differential on a per-packet basis in
               order to minimize packet length.
     How should the difference between the options provided by the trace
     control interface options and the Etrace specification be reconciled?
     What controls select the data compression form?

5. "Rc. Terms hart/core are not always consistently used." Please address
    the inconsistency.

6. "RC. It is divider so down "  Please update the description.