riscv-non-isa/riscv-nexus-trace #55: Feedback on RISC-V Trace Control Interface Specification - chapter 11

ved-rivos, 2024-04-14:
Section 11.4 claims that version 0 implies an original version of this specification. The text in this section is misleading as there is no RISC-V ratified version of trace control specification. While there may be many proprietary/custom implementations claiming version 0 being a legacy version of this specification is misleading.

mipsrobert, 2024-05-02:
Closing with the following changes:
1.	DONE: trAtbBridgeID is NOT a replacement of IDs inside of trace packets. Entire RISC-V trace stream (from several cores …) is encapsulated in ATB formatted stream. ID of that stream is defined by ‘trAtbBridgeID’. It was a bit re-worded to make sure it is NOT replacement and that entire stream is encapsulated.
2.	NOT DONE: This is always optional. It is WARL and may be hard-coded to 1 (or 0). This spec does not define how ATB bridge is connected to ATB sub-system.
3.	DONE: It is not a requirement, but a reasonable suggestion. Also, when component is released from reset, all other fields (except reset bit) should be kept the same. Reading back allows trace tool to cache/shadow this register value and only change some bits and write it. It is much more efficient to write several values instead of doing more costly read-modify-write. Also reading is not necessary after future resets. ACTION: Used words ‘Optionally save’ and ‘shadow/cache’ to clarify intentions.
4.	DONE: Agree. Word ‘legacy’ is changed to ‘pre-release/initial’ (in entire document). I am aware about several independent implementations – all of them are claiming ‘version 0’ – this was discussed in TG.

closed 2024-05-02 by mipsrobert

ved-rivos, 2024-05-14:
In section 11.2 should the bullet 3 - "Optionally save tr??Control...." not be after the current bullet 4 i.e. "Release from reset...". Why is the recommendation to save the value when the unit was held in reset and not when the unit was released from reset?