riscv-non-isa/riscv-nexus-trace #56: PIB, RAM, and ATB alignment synchronization

ved-rivos, 2024-04-14:
PIB, RAM, and ATB have a control field defined to enable generation of alignment synchronization packets. However, these components do not have a trace format (E-trace or N-trace) configuration. As the alignment synchronization sequence is protocol specific, should there should be equivalent of trTeFormat control provided for these components.

mipsrobert, 2024-05-02:
Closing with the following comment:

NOT DONE: It was not a intention to mix and match different trace protocols, but provide control spec to be identical/re-used by both - at least not now. Such things can be added as future revision/addition (if needed). ARC was against specifying future things, so why do it here. As interfaces between trace components are not defined (by this spec …), nothing prevents ‘trTeFormat’ to be passed from encoder to funnel and sinks.

closed 2024-05-02 by mipsrobert

ved-rivos, 2024-05-14:
What is the motivation for making trTeFormat as WARL and not RO?