riscv-non-isa/riscv-nexus-trace #59: 【Discuss】About NexRv Tool

micreven, 2024-05-31:
Hi All，

When I Use the NexRv Tool，I found it expect the first “ProgTraceSync” msg’s “full addr” field match the code’s first PC.
But I think when I enable the trace encoder, the Hart may not start yet. 

I wonder if I used the tool wrong ?

Any help would be greatly appreciated!
 
Best regards,
Micreven

mipsrobert, 2024-06-12:
NexRv tool is based on sequence of PCs which retired. First PC is reported in very first message. Again all retired instructions are reported on ingress port, so if hart is not started yet, there is nothing to report.

micreven, 2024-06-12:
Hi Robert，

Thanks for your explanation.

But as the SPEC said，the “ProgTraceSync” message should be generated when 'trTeInstTracing' changed from 0 to 1, whenever the hart is start or not.

How could the trace encoder generate the "full-addr" field for the "ProgTraceSync" msg, when hart not start and 'trTeInstTracing' changed from 0 to 1. I think this moment, the ingress port can't give any information for Trace Encoder.

Best regards,
Micreven

mipsrobert, 2024-06-12:
Obviously trace cannot be reported if ingress port is not providing any retired instructions. But it should report message at nearest opportunity. Technically you may have a hart being held in reset (for long time) and enable trace while hart is held in reset. Nothing is reported until hart will retire instruction.

micreven, 2024-06-27:
Hi Robert,
Thanks for your explanation. 
Now I got it.
And do you prepare to release the next version of NexRv Tool ? It seems like cant match the newest SPEC.

Thanks

Best regards,
Micreven

mipsrobert, 2024-08-14:
Now the spec is officially Frozen. I will wait for end of public review (30 days from today).