riscv-non-isa/riscv-nexus-trace #61: trTeCompjControl.trTeCompSFunction reg. definition in <trace control interface spec>

zhangdujiao, 2024-07-02:
the value of trTeCompjControl.trTeCompSFunction seems to be repeated at value=6 and 7, they both represent result always true?

mipsrobert, 2024-08-14:
I was waiting for official Freeze notification. This will be considered as Public Review comment.
@IainCRobertson may provide answer to this.

IainCRobertson, 2024-08-14:
Sorry, I thought I'd responded to this.  Yes, they both ignore the actual input value, and are both "always true".  But additionally, code 6 reconfigures the primary comparator to use trTeCompPMatch as a match, so the primary match becomes 

(input & trTeCompSMatch) <pfunction operator> (trTeCompPMatch & trTeCompSMatch).

What prompted this question?  I'm struggling to see how this isn't clear from the spec.

zhangdujiao, 2024-08-16:
> (input & trTeCompSMatch) (trTeCompPMatch & trTeCompSMatch).

thanks Iain, no problem for me :)

closed 2024-08-16 by zhangdujiao