riscv-non-isa/riscv-nexus-trace #63: The discovery method and principle of trace components

zhangdujiao, 2024-08-04:
From <RISC-V Trace Control Interface Specification>, 4.1. Trace Components
"Connections between different trace components must be also defined. Future versions of this specification may allow a single base address to be sufficient to access and discover all trace components in the system."
How to understand this discovery method? 
Will each trace component have a base address of next component? like the debugger knows the 1st address of DM and next DM can be provided by the nextdm register of 1st DM, from the debug spec. 
If we have multiple TE and multiple Sink in SoC, how to discover them all? 
many thanks!

zhangdujiao, 2024-08-08:
can anyone support the help?

mipsrobert, 2024-08-14:
Hello - sorry for delay. I was waiting for official freeze acceptance.

There is (on purpose) no way to discover all components as said at the end of Chapter 4.1.

This 'future ...' section means, that future version may provide registers allowing to find entire 'tree' starting from single 'leaf'. For example each encoder may have a registers which provide base addresses of next encoder and funnel and so on.

NOTE: This is a bit more complicated that 'nextdm' as we need to describe a 'tree' and connections are important. If you have 10 encoders and 3 funnels, you may not have an idea how these are connected together.

closed 2024-08-14 by mipsrobert