riscv-non-isa/riscv-nexus-trace #65: B-TYPE to be clarified for some SYNC messages

mipsrobert, 2024-08-21:
Markus brought (in side-email thread ...) an issue of some SYNC messages (periodic or notifications) which may be generated in the middle of linear code. Spec was (silently ...) assuming, that B-TYPE=0 (as Interrupt or Exception are clearly not correct) will be used, But B-TYPE=0 is currently described as "Indirect control flow change (jump, call or return).".

mipsrobert, 2024-08-21:
Here is proposed solution:
1. Add a new paragraph in 8.4 detailing how some events (periodic sync, notifications, ...) may generate Synchronizing message in the middle of linear code.
2. Update B-TYPE=0 for above.
This solution will NOT break any existing implementation (as 0 was the only possible choice) - it will just clarify that case.

mipsrobert, 2024-08-22:
And also add this at the of of every Sync message.

This message may be also generated (with B-TYPE=0) on linear code for certain synchronization events as described in <<Synchronizing Messages>> chapter.

The <<Synchronizing Messages>> denotes link to chapter 8.4.

jgamoneda, 2024-08-23:
Yes, I agree with these changes

markuslauterbach, 2024-08-23:
Yes, I also agree with these changes

closed 2024-09-17 by mipsrobert

markuslauterbach, 2024-09-17:
Hi @mipsrobert 

regarding this new passage:

> Some synchronizing messages not related to code being executed (periodic, notifications etc.) may be emitted between indirect jumps. In such a case field B-TYPE=0 will be emitted, but it will not mean indirect flow change. 

I think the reference to "**emitted between indirect jumps**" is confusing, because this does _not_ cover all scenarios for which this rule is relevant. This leads to the false assumption that this rule is only applicable if there is an indirect jump before and after the sync message. However, the rule should also apply to scenarios in which there are no indirect jumps at all (or at least not "near" the sync message).

Would it make sense that the rule does not refer to indirect jumps in particular, but only to (any kind of) control flow changes in general?

Maybe rephrase to: "Some synchronizing messages not related to code being executed (periodic, notifications etc.) may
be emitted even though there is currently no control flow change. In such a case, field B-TYPE=0 will be emitted."?

Regards, Markus

mipsrobert, 2024-09-17:
Hi @markuslauterbach 
I explained this to @jgamoneda  as well ...

Our original text has a term 'linear code'. But I did NOT use it from two reasons:

1. This term was 'protested' (by Ved) during ARC review as 'unknown' in RISC-V ...
2. It was NOT exactly correct.

Why I think 'NOT exactly correct'?

Let's assume you have a program, which DOES not have any indirect jumps (call/ret ...) at all.

To give you practical example, let's consider a memcpy for very large area. In this case you HAVE 'control transfers' (conditional branch) but you do NOT have any packets with address for very long time.

All message you will get will be messages with I-CNT and HIST. Correct? So, if you want periodic sync, you must emit message with SYNC and B-TYPE=0 (=indirect!). That's why I said, 'between indirect jumps'.

And even more - if you have a "while (1) {}"code with only one instruction being 'direct jump at itself', then all you have is 'control transfer' instructions ONLY. Saying 'linear code' does not apply here.

Agree?

mipsrobert, 2024-09-17:
@markuslauterbach - thinking a bit more ... I could add above two examples, to clarify more, but IMO everyone reasonable will 'invent' these. 

Biggest 'problem' we wanted to clarify is, that 'indirect' (B-TYPE=0) can be used when there is no indirect jumps and you may still be forced to emit it ...

Maybe 'between indirect' is also not exactly correct ...

I could use 'in place when there is no indirect' instead 'between indirect', but let's not make disruption in the ratification flow. I am really hoping for this to be ratified before RISC-V Summit (Oct 21-st ...).

And finally - apologies - we could discuss this exact wording before. I was thinking 'linear code' was 100% agreed, but I only came to that idea, that it is not exactly correct after I copy and paste this. Me and @jgamoneda discussed it (as I really wanted to act fast ...) and make next move ...

markuslauterbach, 2024-09-17:
Hi @mipsrobert 

I agree that "linear code" may not be precise enough either. However, please note that my alternate suggested wording in my previous message does not contain the term 'linear code', but rather "no control flow change" (not sure if that is better). Essentially we just want to say that the original "criteria" for the messages get softened/ignored in certain scenarios.

But if there was a decision against the use of the phrase 'linear code', then why did this commit add four sentences with 'linear code' in them?

> All message you will get will be messages with I-CNT and HIST. Correct? So, if you want periodic sync, you must emit message with SYNC and B-TYPE=0 (=indirect!). That's why I said, 'between indirect jumps'.

Why does B-TYPE=0 mean "=indirect" in this particular example? Wouldn't it rather mean the alternate B-TYPE meaning "in a synchronizing message not related to code execution.", which means the example is _not_ related to indirect branches?

My interpretation of the wording "synchronizing messages [...] may be emitted **between** indirect jumps." would be the following example (for simplicity, let's assume each indirect branch jumps to the next instruction):

_jr x5    // <--- indirect jump (generates normal (non-sync) branch message)
nop
nop    // <--- Periodic SYNC message, which gets emitted "**between (two actual) indirect jumps**"
nop
jr x6    // <--- indirect jump (generates normal (non-sync) branch message)_

If it's already too late to change anything then we can of course also continue this discussion after the ratification. Since the change was made yesterday, I thought that I would share my two cents about it.

Best regards, Markus

mipsrobert, 2024-09-17:
I think we agree, but we (both) try to put too much attention to specific words and maybe try to make things too perfect ...

By saying  "**B-TYPE=0 (=indirect)**" I was referring to (original ...) meaning of B-TYPE=0, what WAS indirect (only). Now it was softened and existing text just says 'B-TYPE=0' (without referring 'indirect').

By saying 'between indirect' I did mean in a sequence of PC flow, not in memory order ...

IMO this is what you wanted to change (allow extra meaning for B-TYPE=0) and we changed it.

I fully understand your intentions for full clarity, but let's move on ...

@jgamoneda - maybe you can put some closing words for this discussion?

jaygamoneda, 2024-09-17:
The changes made to the spec address the main concern highlighted in issue #65, that (periodic, notification, etc) SYNC message may be generated at any time, not just change of flows. Also updating that B-Type=0 is a valid for this scenario. 

Current SPEC has this wording:
> Some synchronizing messages not related to code being executed (periodic, notifications etc.) may be emitted between indirect jumps. In such a case field B-TYPE=0 will be emitted, but it will not mean indirect flow change.

Markus is suggesting
> Some synchronizing messages not related to code being executed (periodic, notifications etc.) may
be emitted even though there is currently no control flow change. In such a case, field B-TYPE=0 will be emitted.

I am fine with either wording as I interpret them the same, "Branch SYNC" message can be generated during "linear code" execution.

I would not object to use Markus more generic wording, but I don't think it is necessary to change.

mipsrobert, 2024-09-17:
Thanks you both! I think this text:

**Some synchronizing messages not related to code being executed (periodic, notifications etc.) may
be emitted even though there is currently no control flow change. In such a case, field B-TYPE=0 will be emitted.**

is better, but it does NOT cover a case, when periodic SYNC is generated at DIRECT branch (or jump), which is 'control flow change'.

**Simply saying, periodic (or notification or etc.) SYNC may happen at ANY instruction in the PC flow.**

Let's not discuss this more and keep existing text, from several reasons.

markuslauterbach, 2024-09-18:
Okay let's abort the discussion for now. I guess only time and experience with the spec in the field will tell which passages people are struggling with.

jiangshuaili, 2025-11-21:
[image]

There is a little description error.