riscv-non-isa/riscv-nexus-trace #73: 64-bit counter overflow

Graf-Zahl, 2024-10-23:
"(even with 1ps resolution, 64-bit counters will overflow in about 584 years)"

I think the overflow is calculated for 1ns instead of 1ps.

https://github.com/riscv-non-isa/tg-nexus-trace/blob/main/docs/RISC-V-N-Trace.adoc?plain=1#L1476

mipsrobert, 2024-10-28:
Yes. I am glad someone is reading spec carefully :-). It should be 1ns as it corresponds to 1GHz clock. But even 0.584 years is a lot of trace time. I will refrain form making any changes and wait for formal ratification before doing any changes. I will keep this ticked open until it is addressed (and fixed).

mipsrobert, 2025-02-26:
Spec is now ratified. But let's keep it open - process to make not essential 'typo changes' is not well defined 👎