riscv-non-isa/riscv-nexus-trace #77: trRamRP register purpose

aroca-smd, 2025-02-14:
Hi everyone,

My question is about the trRamRP register. As I understand it, in SMEM mode the trRamRP pointer is defined as optional because the trace data is not read through the control interface, but rather directly from memory through SBA or a similar mechanism (which I think makes sense), but I don't see why would it be required in SRAM mode.

In my mind having a RP implies allowing random access to the trace SRAM, but shouldn't it behave just like a FIFO where you only ever read the head?  I can't think of a situation where the user would want to set a RP because if data isn't read in order it breaks the trace flow.

I guess that being a WARL register technically gives the freedom of treating the address of the head as the only legal value but it seems a little janky.

I would appreciate any clarification on this, thanks!

mipsrobert, 2025-02-18:
Sorry for late reply ...

RP is needed in SRAM mode from two reasons:
* to know where is start of trace.
* to allow trace tool to NOT read entire trace every time.

Imagine a situation, where you have small SRAM buffer and trace is constantly wrapping (changing WP). You stop the trace. Then WP is in the middle. Here is hypothetical trace buffer (with 8 bytes) and trace which generated 1239 bytes (byte #0..#1238).
Certainly you only have last 8 bytes in memory as follows:

Location Byte#
ram[0]:    1234
ram[1]:    1235
ram[2]:    1236
ram[3]:    1237
ram[4]:    1238    <- this is NEWEST (last) trace byte
ram[5]:    1231    <- WP points here (has value 4). This is OLDEST (first available ...) trace byte
ram[6]:    1232
ram[7]:    1233

Trace tool must read bytes ram[5] .. ram[7] and then bytes ram[0] ... ram[4]. As follows:

RP=5 (value of WP)
read 'data' 3 times and save them into local buffer (from start)
RP=0 (to continue reading from beginning)
read 'data' 5 times and save them into local buffer (from position=3)

Please note, that same 'technique' is used to read big buffer (from SMEM) and also from internal memory inside of trace probes.
Read from 'WP' till end, then read from 0 till 'WP-1'.
If trace did NOT wrap, you only read from 0 till 'WP-1'.

It would be possible to read from ram[0]-ram[7], but then trace tool would need to analyze a buffer from the middle and wrap around or do some 'swapping' after reading. This would be a nightmare for 1GB buffer.

@aroca-smd  - I hope it clarifies it - please close the issue if you are happy now.

aroca-smd, 2025-02-20:
Okay then, it is just that it seemed a little fuzzy to me to who (hardware or trace tool) had the responsability to keep the pointers updated. But I see that the hardware can usually do it on its own but must allow for the trace tool to overwrite it.

Thanks, issue closed :)

closed 2025-02-20 by aroca-smd

mipsrobert, 2025-02-20:
I agree, that if RP=WP (at trace stop, you may never set it and read entire trace).
But you will NOT be able to read partial trace.
Spec says (in Miminal implementation chapter):

Register trRamRPLow must accept any 32-bit aligned value in inclusive range < 0 .. trRamLimitLow >.