riscv-non-isa/riscv-nexus-trace #78: Open Source RTL Implementation?

omerkarslioglu, 2025-02-15:
Hi I want to add n-tracer to my RV core. Is there any RTL implementation of n-tracer? And how should I start to design n-tracer for my RV core? Could you please show a road map for it?

omerkarslioglu, 2025-02-18:
Thanks Mr. Robert, it is helpful.
"Then make sure you send proper messages in proper moment."

Is that means encoding ingressport data to nexus style messages?

@mipsrobert

mipsrobert, 2025-02-18:
Yes. For example when you have exception reported in ingress port (I do not remember how ...) you need to emit a message including accumulated ICNT and HIST and proper SYNC field. Then next PC (from ingress port ...) will be exception handler address.

Please close this issue once you are happy ...

omerkarslioglu, 2025-02-20:
Hi Mr. @mipsrobert again,

I am trying to implementing nexus encoder RTL now, I am getting reference code from "10.2. Pseudo-code of Simple N-Trace Encoder" when I try to understand minimal implementation. Is that pseudo code ensure requirements of n-trace spec? 

I am not sure, wheter it is enough or not to get reference.

mipsrobert, 2025-02-20:
It is just for illustration only, but is most likely accurate. True encoder is inside of repo. It was run on many test programs and IS generating correct trace.

closed 2025-02-20 by mipsrobert