riscv-non-isa/riscv-nexus-trace #79: A missunderstandable condition in Pseudo code from the spesification

omerkarslioglu, 2025-02-24:
```C
    uint32_t prevICNT = encoICNT; // In case ICNT will overflow now, we need to emit previous value ...
    if (InfoIsSize32(info)) encoICNT += 2; else encoICNT += 1;

    // Determine type of message (only if this is branch or indirect ...)
    if (InfoIsBranch(info)) 
    {
        if (enco_opt_branch_history) // eger branch history aktifse
        {
            // Update branch history buffer (add least significant bit)
            if (InfoIsBranchTaken(info))
                encoHIST = (encoHIST << 1) | 1; // Mark branch as taken
            else
                encoHIST = (encoHIST << 1) | 0; // Mark branch as not-taken
        }
        else
        {
            if (InfoIsBranchTaken(info))
                encoNextEmit = NEXUS_TCODE_DirectBranch;    // Emit destination address (next retired)
            else
                ;   // Not-taken branch is considered as linear instruction
        }
    }
    else if (InfoIsIndirect(info))
    {
        if (enco_opt_branch_history)
            encoNextEmit = NEXUS_TCODE_IndirectBranchHist; // Emit destination address (next retired)
        else
            encoNextEmit = NEXUS_TCODE_IndirectBranch; // Emit destination address (next retired)
    }
```

I think ``enco_opt_branch_history`` represents HTM mode.

And when I look at the explanation of HTM, only direct conditional branches updates history buffer. But in the above (pseudo code), all direct branch could change the history buffer. (eg. "jal" can also change history buffer according to pseudo code.)

Is there any mistakes in the pseudo code?

aroca-smd, 2025-02-24:
Hi, 

As I see it, just like you said HTM mode only takes into account conditional branches, but you have to keep in mind that the Trace Encoder works with the data provided by the Ingress Port, which would report TCODE = 0 for a JAL, meaning it does not consider a JAL as a "special" instruction as it can be statically inferred.

What I mean is that N-Trace (and I'm pretty sure E-Trace as well) doesn't really distinguish between a normal linear instruction and a direct unconditional jump, so it is implied that in the pseudocode only refers to conditional direct branches when talking about direct branches.

> <html><body>

Instr. | Desc. | ITYPE (3 bits case, no Implicit Return optimization)
-- | -- | --
JAL rd | Any direct jump/call. | 0 = No special type
JALR rd, rs | Any indirect jump/call. | 6 = Indirect jump (with or without linkage)
C.J or C.JAL | C extension has direct jump/calls only. | 0 = No special type
CM.JT | Defined by Zcmt extension. | 0 = No special type
CM.JALT | Defined by Zcmt extension. | 0 = No special type
CM.POPRET* | Defined by Zcmp extension. | 6 = Indirect jump (with or without linkage)

(N-Trace Specification Chapter 2)

closed 2025-02-24 by omerkarslioglu

omerkarslioglu, 2025-02-24:
```
Not-taken direct conditional branches and direct unconditional jumps increment I-CNT
but do not generate any trace. Direct unconditional jumps change the PC to the
destination address of such a jump. The I-CNT enables determination of the PC of the last
instruction in the code block(s).
```

I saw also above text in the spec.

mipsrobert, 2025-02-26:
Sorry I missed this. Solution is in terminology 'branch' is ALWAYS conditional (B* instructions) in RISC-V. Every other flow changes are called 'jump' (J* instructions) are either direct or indirect. This is described in ISA manual and both Trace Control and N-Trace specs. Here is most relevant part:

===
The terminology Indirect Branch as used by the IEEE-5001 Nexus Standard may lead to
confusion, given that the RISC-V ISA exclusively permits direct conditional branches,
which are always relative. Furthermore, the RISC-V ISA makes a distinction between 'jump'
(unconditional flow change) and 'branch' (conditional flow change), a differentiation not
observed in Nexus terminology, where any flow change, including exceptions and
interrupts, is uniformly referred to as a 'branch'. This specification employs the terms
'branch' and 'jump' as defined by RISC-V ISA.
===